rusefi/hellen154hyundai #2: What is this alternator wire

rusefillc, 2021-09-12:
Is that input? output? do we simply ignore it for now?

[image]

rusefillc, 2021-12-17:
@ElDominio do we need to do anything here or is it running fine using whatever fail safe more?

ElDominio, 2021-12-17:
I mean, the car didn't light on fire today, so I suppose so lol

I'll confirm tomorrow but it's probably just working

rusefillc, 2022-08-23:
@ElDominio are we using that wire? Anything to do here?

ElDominio, 2022-08-23:
I have not attempted using it. Alternator locks at 14v when not connected, maybe connecting and using it may give us a few hp? 🤷🏿‍♂️ I've never done alternator control

ElDominio, 2022-12-03:
Close

rusefillc, 2022-12-03:
c'mon, this one we shall figure out eventually to have the proper smart control strategy