rusefi/hellen154hyundai #17: Review

rusefillc, 2021-09-14:
Wiring diagrams https://github.com/rusefi/rusefi/wiki/Hyundai-Genesis-Coupe

Pinout https://docs.google.com/spreadsheets/d/14IxjitlEFrC2bSp0IRdNRyXZondlvw2CmJ3Q9DUdvyo

andreika-git, 2021-09-14:
input module: +5VA should be connected to the external net:
[image]

andreika-git, 2021-09-14:
TPS1/TPS2/PPS1/PPS2 pullup/pulldown inconsistency:
[image]

andreika-git, 2021-09-14:
Warning: these are not hardware PWM outputs:
[image]
It may affect PWM performance because currently the firmware uses hw pwm for ETB controller...

mck1117, 2021-09-14:
is the wastegate ETB-style? If so, it needs to be on PWM too.  If not, it's not a big deal as it's slower PWM for a solenoid.

andreika-git, 2021-09-14:
MCU-0.4 is used but currently we have 0.5 revision:
[image]

rusefillc, 2021-09-14:
@mck1117 yes wastegate is DC motor

@andreika-git @mck1117 need help with more explicit recommendation where to connect

andreika-git, 2021-09-14:
For example, we could use OUT_PWM2,3 for ETB and OUT_PWM4,5 for WASTEGATE. Solenoid outputs aren't used anyway (just free pads on the pcb), so they are available.

rusefillc, 2021-09-16:
#21 created

rusefillc, 2021-09-16:
@andreika-git is the board good enough now with recent changes?

@mck1117 is there a chance for you to review in the next 96 hours or shall I order?

rusefillc, 2021-09-20:
ordered!

closed 2021-09-20 by rusefillc