rusefi/hellen154hyundai #19: +5VA should be connected to the external net

rusefillc, 2021-09-14:


Qwerty-OFF, 2021-09-14:
@andreika-git 
Aren't these pins connected?
[image]
[image]

andreika-git, 2021-09-15:
Yes, it's connected inside the module, but it also has a TVS on that net:
[image]
The original idea was to feed +5VA to the ECU connector through the input module (and through the TVS) - to suppress higher voltage spikes (for instance, coming from VR-sensors) and make the analog supply less noisy. When the TVS is bypassed by connecting directly to the mcu's +5VA source, then it's less efficient.
Of course it's not a requirement and may not be needed for this board, so it's up to you to decide...

Qwerty-OFF, 2021-09-15:
Good idea, but not obvious. It looks like we need to write about it here https://github.com/andreika-git/hellen-one/wiki/module-input and maybe someday when updating input module, rename this contacts to V5A_IN and V5A_OUT.

I connected external net to this pin
[image]

closed 2021-09-15 by Qwerty-OFF