rusefi/hellen154hyundai #35: five LEDs are dimming with IGN off and no laptop

rusefillc, 2021-10-17:
hook ECU to real vehicle

get 4 red and one white LED dim :(

[image]

[image]

rusefillc, 2021-10-17:
@mck1117 @ElDominio FYI

rusefillc, 2022-03-27:
@Qwerty-OFF poke we've got real users asking :)

Qwerty-OFF, 2022-03-28:
Try to remove this:
[image]

Seems that same problem with always-hot main relay diode:
[image]

Solution for next rev:
[image]

Qwerty-OFF, 2022-08-22:
Is it help https://github.com/rusefi/hellen154hyundai/issues/35#issuecomment-1081100280?
If so, close the ticket, fixed.

Qwerty-OFF, 2023-04-17:
@rusefillc проблема еще есть?

closed 2024-02-21 by Qwerty-OFF