rusefi/hellen154hyundai #42: Pin mappings non-existent

ElDominio, 2021-12-08:
Missing some output pins in firmware

No VVT:
[image]

No boost control for solenoid cars
[image]

A/C is hardware on 2010-2012, CAN on 2013+. Current test car is 2013. If CAN messages are needed, I can supply them
A/C pins for 2010 aren't defined either

Power steering input is not defined, I can't see it in the pinout either, was it not added?

I'll add more as I find more things

mck1117, 2021-12-08:
What does the power steering input do?

ElDominio, 2021-12-08:
Shows when PS pressure go over a threshold (it's a switch at idkhowmany psi) to raise idle a bit, so it doesn't get cramped. Not the most important of pins.

rusefillc, 2021-12-08:
@ElDominio both VVT added, fan low added, WG solenoid added - it's yours to test

todo PS switch a bit later

rusefillc, 2022-08-23:
Closing for now

closed 2022-08-23 by rusefillc