rusefi/hellen154hyundai #44: Document Aux IO

rusefillc, 2021-12-16:
flex
o2 sensor
boost control
fuel pressure
a nitrous solenoid?

rusefillc, 2021-12-16:
[image]

[image]

rusefillc, 2021-12-16:
Looks like in order to get aux analog input we should move CAM2 input from analog-capable AUX4 to SENS2 maybe?

rusefillc, 2021-12-16:
@ElDominio
TL,DR analog inputs is hardest we can MAYEB squeeze in if we add a jumper wire on the board let me know if needed. we use OEM oil pressure on the board which we do not use, your call if you are ready to start repurposing those channels which do not really have software support at the moment

we have a few logic level outputs and we have proto area to make those into full blown high current outputs. it sounds like you are interested in this as an improvement in next revision

flex input being digital input should be possible just need to test

ElDominio, 2021-12-16:
The thing is most clients are interested in adding stuff later down the road, so at least on the PNP LinkECU boards I had designed, I added a 12pin Molex MiniFit to provide for additional IO. The test car currently has an additional fuel pump, has flex sensor, a WB02 and an extra switch wired into this.

https://www.alphaxpr.com/collections/alphax-line/products/genesis-coupe-2-0t-pnp-harness-for-linkecu on the left side

closed 2024-02-23 by rusefillc