rusefi/hellen154hyundai #45: Let's remove some jumpers in order to get space on the board

rusefillc, 2021-12-16:
[image]

rusefillc, 2021-12-16:
[image]

[image]

[image]

rusefillc, 2022-08-20:
@ElDominio please review and confirm that those onces here are SOLID

ElDominio, 2022-08-20:
SOLID meaning they work?

If so yes, all of these work

rusefillc, 2022-08-20:
SOLID as "there is no way on earth I would need to mess with those traces mentioned above, those would always be the default pinout as on current board"

ElDominio, 2022-08-20:
Yep, definitely solid then

Qwerty-OFF, 2022-08-23:
Done

closed 2022-08-23 by Qwerty-OFF