rusefi/hellen154hyundai #46: Some digital inputs are using analog inputs

rusefillc, 2021-12-17:
Analog inputs are hot commodity we should free those from digital inputs

[image]

rusefillc, 2021-12-17:
@ElDominio which of these are analog or on/off digital:

ac_on
clutch
brake
ac_pressure

ElDominio, 2021-12-17:
only AC pressure is voltage, others are switched

rusefillc, 2021-12-17:
@ElDominio do you know if those need pull-up, i.e. are those GND/floating on the wire or voltage/GND or else?

ElDominio, 2021-12-17:
Brake is 12/gnd
clutch is gnd/floating
ac i don't clearly remember, i think it's gnd floating since it's only a request
ac pressure is voltage 0-5

Qwerty-OFF, 2022-08-23:
@rusefillc RES1-RES3 is not analog (https://github.com/andreika-git/hellen-one/wiki/module-input). Should I do anything?

rusefillc, 2022-08-23:
We should move Crank/cam1/cam2 to digital-only pins and have current "crank/cam1/cam2" available as analog.

(overall the big picture: we've made a mistake while selecting default crank/cam - those should never be using analog-capable pins)

Qwerty-OFF, 2022-08-23:
Done https://github.com/rusefi/hellen154hyundai/issues/48#issuecomment-1224642038

closed 2022-08-23 by Qwerty-OFF