rusefi/hellen154hyundai #48: Suggested auxiliary pins; and confirmed pins that don't need jumpers

ElDominio, 2021-12-17:
Suggest aux pins:
12v
5v
gnd
flex
boost duty output
nitrous output
some ground switch (so people can put a switch to do something
fuel pressure 0-5v
oil pressure 0-5v

all confirmed working IO
ETB
VVT1/2
all fuel injectors
all ignition outputs
cam/crank
MAP1
CLT
IAT

I did not check MAP2 (could be used as Baro)

ElDominio, 2022-08-20:
to update this list:

recommend using LCSC part number  C477839

[image]

Pins are as follows:
1. 5V source
2. 0-5v p1
3. 0-5v p2
4. digital input p1
5. Ground PWM p1
6. digital input p2
7. 12v source
8. 0-5v p3
9. digital input p3
10. ground pwm p2
11. ground pwm p3
12. sensor ground

Qwerty-OFF, 2022-08-23:
[image]
@ElDominio please check (reverse numbering of contacts is a feature of the footprint)

rusefillc, 2022-08-23:
@Qwerty-OFF: Norman says "Anywhere is fine"

but we have case shape considerations

maybe here?

[image]

rusefillc, 2022-08-23:
PS: new ETB drivers can go on the back side of the PCB I guess if that would help at all?

We might get double sided inputs and outputs modules in the next few days if that would help

ElDominio, 2022-08-23:
Are there pins available to give exclusive use to the pins I stated in the comment? (not shared with anything on main connector, only dedicated for AUX use)

rusefillc, 2022-08-23:
See https://github.com/andreika-git/hellen-one/issues/155

Qwerty-OFF, 2022-08-26:
> recommend using LCSC part number C477839

This is right angle connector, is it possible to place it in the center of the board, and not on the edge? Should I use the same straight connector?

> Are there pins available to give exclusive use to the pins I stated in the comment? (not shared with anything on main connector, only dedicated for AUX use)

AUX_LS1 connected to OUT_WGT on main connector. If need seems possible place additional MOSFET for AUX_LS1.

How much current should the AUX pins support?

rusefillc, 2022-08-26:
let's use VND7/VND14NV04TR "DPAK-3","C155605"

rusefillc, 2022-08-26:
Please add pull-up/pull-down instructions for all aux analog inputs on the back of the PCB

rusefillc, 2022-08-26:
@ElDominio says to use vertical molex connector

Qwerty-OFF, 2022-08-27:
> use vertical molex connector

OK

> pull-up/pull-down instructions

[image]

closed 2022-08-27 by Qwerty-OFF