rusefi/hellen154hyundai #49: rewire and test brake depal

rusefillc, 2021-12-17:
Remove R

Add a jumper wire between connector side of R and 

[image]

[image]

Note added to https://github.com/andreika-git/hellen-one/wiki/module-input

ElDominio, 2021-12-18:
[image]

brake pedal i can't check if it's working or not since it's crossed out
[pressingbrake.zip](https://github.com/rusefi/hellen154hyundai/files/7738542/pressingbrake.zip)

pressingbrake is a log of me pressing brake pedal

mck1117, 2021-12-18:
right click dash -> load/save -> load dashboard -> pick default in dropdown (should be picked already) -> OK

ElDominio, 2021-12-18:
kk, brake works

didn't know i needed to reload the dashboard lol

mck1117, 2021-12-18:
We changed some stuff recently that necessitated a one-time dashboard reload.  Doesn't happen on new projects.

ElDominio, 2021-12-18:
Clutch input works on newest firmware

rusefillc, 2022-08-20:
Sounds like we are happy and can close @ElDominio ?

ElDominio, 2022-08-20:
Yessir, good to close

closed 2022-08-20 by rusefillc