rusefi/hellen154hyundai #50: relay output for 2nd fuel pump

rusefillc, 2021-12-18:
We want to active 2nd fuel pump relay  when at 4kRPM OR 15psi - it would be Lua since combined logic

open question where to we add that wire?

rusefillc, 2021-12-18:
We have https://github.com/andreika-git/hellen-one/wiki/module-output

we have a lot of outputs available

[image]

rusefillc, 2021-12-18:
@ElDominio let's use P69 I will now make it available in TS

[image]

Qwerty-OFF, 2022-08-23:
@rusefillc should I do anything?

ElDominio, 2022-08-23:
This is also fine if routed to aux connector.

rusefillc, 2022-08-23:
we can add up to aux 10 outputs it's the question of level of effort

rusefillc, 2022-08-26:
Sounds like one of low side aux ouputs

closed 2022-08-26 by Qwerty-OFF