rusefi/hellen154hyundai #51: WG solenoid wiring

rusefillc, 2021-12-18:
Any side of R30 could be used to add an extra wire for aftermarket WG solenoid

[image]

rusefillc, 2022-08-23:
@ElDominio do we need this open or time to close?

ElDominio, 2022-08-23:
Will the aux port (molex) I mentioned be implemented? If so, it would obviate this since we can use one of those PWM outputs

rusefillc, 2022-08-23:
Do we have molex ticket for this hellen154hyundai?

ElDominio, 2022-08-23:
#48

Qwerty-OFF, 2022-08-23:
Done
[image]

rusefillc, 2022-08-26:
let's have R30 DNP

rusefillc, 2022-08-26:
Let's add schematics note next to main connector just to clarify that on main connector it's "OEM BK1" 

We want same trace/output to be routed to aux connector for BK2 people to use it if desired without adding a wire into OEM harness

rusefillc, 2022-08-26:
STATUS UPDATE: let's actually DETACH OUT_WG_BK1 from aux connector

please let's have all aux connector low sides have nothing to do with main connector outputs

overall the idea is to not confuse user support between BK1 and BK2, much safer to have consistent story and just ignore unused OUT_WG_BK1 on BK2 vehicles

Qwerty-OFF, 2022-08-27:
Done

closed 2022-08-27 by Qwerty-OFF