rusefi/hellen154hyundai #54: AC Request not working

ElDominio, 2022-04-09:
I tried on a BK1, seems the pin is floating? On "default" it thought it was on (apprently no voltage at pin) and I activated the pullup and it turned off. Physical button did no change.

It also activates while starting? Tune and log on rusefi online

rusefillc, 2022-04-09:
I assume
https://rusefi.com/online/view.php?msq=939
and
https://rusefi.com/online/view.php?log=737 https://rusefi.com/online/view.php?log=736

rusefillc, 2022-04-09:
just a reminder about rusEFI console "live data" view

let me try that "live data" for AC control on my NB2 tomorrow

ElDominio, 2022-04-10:
Correct, let me know so I can test further

ElDominio, 2022-06-23:
Tested this, no change on anything on liveview

rusefillc, 2022-06-24:
please update to latest firmware with "view" in .ini

and post a fresh log reproducing the problem

ElDominio, 2022-06-28:
[2022-06-28_14.41.19.mlg.zip](https://github.com/rusefi/hellen154hyundai/files/9004930/2022-06-28_14.41.19.mlg.zip)

no change is detected, im pressing the button every which way

rusefillc, 2022-07-02:
RPM=0 means "engine is too slow to enable A/C". This log works as expected?

Note "view->AC" dialog - i assume it would show which condition is YES while all other conditions are NO?

[image]

[image]

ElDominio, 2022-07-02:
I was pressing the a/c button on/off like 10 times and there is no change anywhere, I should at least be able to see a change of state on A/C request even if the compressor won't engage because of software conditions

rusefillc, 2022-07-02:
Ohhhh

Well, we've made progress i finally understand the problem statement!

rusefillc, 2022-07-02:
@ElDominio do we believe that A/C request is pin 57 of 60 pin connector?

[image]

@Qwerty-OFF what do we have for pull-up/pull-down on that RES input?

[image]

[image]

rusefillc, 2022-07-02:
@Qwerty-OFF says's 680k to 5V

that sounds too low to me

please add 10K resistor between "inner" side of R57 jumper and 5v point anywhere on the prototying area

ElDominio, 2022-07-05:
Pin 57 measures 4.7ish volts with AC OFF, 11.3ishvolts with AC ON

rusefillc, 2022-07-06:
@Qwerty-OFF would a voltage divider get values into stm32 TTL range?

Qwerty-OFF, 2022-07-06:
@rusefillc its possible if needed

ElDominio, 2022-07-06:
Will check tomorrow, was tested with rusefi.

Qwerty-OFF, 2022-07-07:
> R57
Typo, R58, not R57...

It looks like it's a high side button. Try to remove pullup and place pulldown.

ElDominio, 2022-07-07:
No ecu results:

0v when off
12v when on

I assume I had left pullup on RusEFI for some reason; but disabling it still does not make the behavior work as expected

rusefillc, 2022-07-08:
@ElDominio please clarify if 10K was added anywhere maybe with a picture

please note the correction in where 10K needs to be added it's now R58 for pin 57

ElDominio, 2022-07-15:
Completely forgot, writing this message to remind myself to do it while I have a BK1 at the shop

Guelphgenesis666, 2022-08-01:
Any one have any pics of where to place 5k resistor between r58 and 5v point

rusefillc, 2022-08-01:
@Guelphgenesis666 please use proto area

just check that pads are not connected to each other since those are paired or tripled

[image]

Guelphgenesis666, 2022-08-01:
So just connect resistor to 5v pad and then other end to r58?

[image]<https://user-images.githubusercontent.com/48498823/182100504-6641bc7c-3b87-4ea4-be9d-fae2e0fc11d8.png>

rusefillc, 2022-08-01:
yep

maybe post a photo before turning power back on :)

rusefillc, 2022-08-01:
photo above is the correct way to do "please add 10K resistor between "inner" side of R57 jumper and 5v point anywhere on the prototying area" as requested at https://github.com/rusefi/hellen154hyundai/issues/54#issuecomment-1172979692

it looks like https://github.com/rusefi/hellen154hyundai/issues/54#issuecomment-1178231342 is suggesting to do pull-down instead which would mean resistor to GND not +5v

overall I am pretty confused for sure but you are pretty close :)

@qwerty-off FYI

Guelphgenesis666, 2022-08-01:
> @ElDominio please clarify if 10K was added anywhere maybe with a picture
> 
> please note the correction in where 10K needs to be added it's now R58 for pin 57

This says r58 . And that is a 10k resistor I’m
Using. Do I need to move it to here instead? 
[image]

Guelphgenesis666, 2022-08-01:
[image]

rusefillc, 2022-08-01:
@Guelphgenesis666 this github here is not the best chatting platform, note the missing images. with too much gaps between communications I am forgetting context. we need to find a different way to get in touch when we both have time to get more efficient.

Guelphgenesis666, 2022-08-02:
Here’s my email If you can send me a msg. J.p.ouellette@hotmail.com

rusefillc, 2022-08-05:
@ElDominio please test with 10K between lower side of R58 and GND, half like the picture

Guelphgenesis666, 2022-08-07:
I added the 10 K resistor between the lower side of R58, And ground, Using pulldown AC switch mode, I am now seeing the AC request when I hit the ac button. Just going to Solder The resistor in place so I can reassemble the ECU and go test drive. I’ll let you guys know in an hour or so
[image]

Guelphgenesis666, 2022-08-07:
[image]

Guelphgenesis666, 2022-08-07:
No power to the ac compressor clutch. Tuner studio shows the button being pressed and says ac is on, however there’s no power to the compressor clutch

rusefillc, 2022-08-07:
so AC request is fixed?

@ElDominio had BK2 A/C working with CAN bus A/C request so BK2 ac compressor clutch is good? are there any differences between BK1 and BK2?

Guelphgenesis666, 2022-08-07:
[image]

rusefillc, 2022-08-07:
@Guelphgenesis666  looks like A/C output not configured?

@ElDominio what should it be for BK1? what should it be for BK2? do you have a link to your working BK2 tune?

[image]

Guelphgenesis666, 2022-08-08:
How do I configure it. What settings do I need?

[image]<https://user-images.githubusercontent.com/48498823/183316153-8cd9fb97-99d8-4039-84e0-a1d9ebac088a.png>

Guelphgenesis666, 2022-08-08:
I set ac relay to “k87 ac relay” . What do I need for ac relay mode

[image]<https://user-images.githubusercontent.com/48498823/183316153-8cd9fb97-99d8-4039-84e0-a1d9ebac088a.png>

rusefillc, 2022-08-08:
default should be good. please power cycle ECU just to make sure setting is applied

ElDominio, 2022-08-09:
Can confirm, resistor jumped from gnd to R58 fixes the issue.

rusefillc, 2022-08-09:
@Qwerty-OFF we would need schematics chagne here

note that only half of units ("first generation BK1") would be using those so we would at least need a comment on the schematics

Qwerty-OFF, 2022-08-23:
Fixed
[image]
[image]

closed 2022-08-23 by Qwerty-OFF