rusefi/hellen154hyundai #55: L6205PD is too weak to run ETB in single channel configuration

rusefillc, 2022-04-12:
we have to move to dual 9201

in the meantime would need to hack current boards to make it match nissan

[image]

Qwerty-OFF, 2022-04-12:
Connect together 45+132 and 60+131
[image]
Cut this (or set to Hi-Z/Input wastegate pins on STM32):
[image]
Connect this:
[image]

rusefillc, 2022-04-13:
Does it have to be diagonal? Is there a way to connect pin 6 to pin 7 and pin 14 to pin 15?

[image]

Qwerty-OFF, 2022-04-13:
It is possible, but in this case the maximum current will be two times lower

rusefillc, 2022-04-14:
See https://github.com/rusefi/hellen154hyundai/issues/56

mck1117, 2022-04-14:
> It is possible, but in this case the maximum current will be two times lower

huh? why? Paralleling the two channels will double the current capacity, and halve the on-resistance (halving the heat!).

Qwerty-OFF, 2022-04-14:
[image]
[image]
Internal current sensor placed only on high-sides:
[image]
and work independent in part A and B, but sum currents from both high-sides of part

mck1117, 2022-04-14:
Ah, I see, the question is whether you do 1A1B + 2A2B vs. 1A2A + 1B2B. I agree, the diagonal way is superior to adjacent pins. However, we still get some of the benefit doing the "adjacent way", since that still reduces Rds_on, while not increasing the current limit. I'm reasonably sure we're hitting the thermal cut rather than the hard current limit.

Guelphgenesis666, 2022-07-17:
For this step: Connect 45/R57 with 132/R45 and 60/R60 with 131/R44.  
do the wires need to extend outside the ecu housing or just connect the wires r57 with r45 and r60 with r44

rusefillc, 2022-07-17:
Nothing goes outside of the ECU

just a jumper wire from PCB to PCB, it's just the picture shows it going out for clarity

I am failing to find my reference photos of when I did the mod myself :( @ElDominio do you have any pictures?

Guelphgenesis666, 2022-07-17:
i added the 2 jumper wires , scratched through the 2 tracer wires , now I just need to connect pcb to pcb. That all that’s needed?

rusefillc, 2022-07-17:
[image]

not the best photo but you can see the blue wire and a short piece of the green wire in the top-left corner just to give you the idea

rusefillc, 2022-07-17:
correct

please post a pic of what you've done

rusefillc, 2022-07-17:
PS: there was no need to scratch, software was taking care of traces

Guelphgenesis666, 2022-07-17:
This step [image0.png]

rusefillc, 2022-07-17:
@Guelphgenesis666 please try again but give it more time to upload the image

Guelphgenesis666, 2022-07-17:
[image]

rusefillc, 2022-08-20:
@Qwerty-OFF please migrate to two TLE9201

Qwerty-OFF, 2022-08-23:
Done
[image]

closed 2022-08-23 by Qwerty-OFF