rusefi/hellen154hyundai #57: Cam settings causes car to lose sync/become inoperable.

ElDominio, 2022-08-09:
Tuning a car, I added CAM sync stuff to enable VVT, the car lost sync and wouldn't even start correctly. Settting cam sync to inactive would still be impossible to start, had to disable injection to clear flood.

I suggest removing first/second half settings and only having a half moon setting, then allowing a +/-360 degree offset
[My Big Turbo Meth 1500cc.zip](https://github.com/rusefi/hellen154hyundai/files/9286092/My.Big.Turbo.Meth.1500cc.zip)

tune and logs attached

rusefillc, 2022-08-09:
it looks like with your current offset your position was too close to 360, and above 360 puts you into negative position

can you substract 360 from offset to get closer to zero position?

[image]

rusefillc, 2022-08-09:
just a reminer that "intake" cam is the important one, that one controls engine phase
the other one which we call "exhaust" should not affect engine phase

ElDominio, 2022-08-09:
[image]

ElDominio, 2022-08-09:
Things we have learned:
[image]

-Having  "maximum cam/vvt sync RPM" to a value slightly above cranking makes the decoder go crazy and start changing values from positive to negative, it needs to be over idle if it is even being used.

-still don't understand what is the difference between "single tooth first/second half" and why can't we just offset crank trigger by 360

-Tunerstudio is not accepting any values in this table:
[image]

rusefillc, 2022-08-09:
@ElDominio i see the table values defect let me fix

rusefillc, 2022-08-09:
https://github.com/rusefi/ts-issues/issues/52

ElDominio, 2022-08-11:
[https://youtu.be/pd6hGU-FxDA]() example of issues. Starts fine once, then after shutting down it will never restart since apparently vvt angle changes sync point

rusefillc, 2022-08-11:
@ElDominio I am just a java developer with pretty limited abilities

being far from the vehicle I need tune which matches video and logs matching video as a starting point to even have a chance to help

ElDominio, 2022-08-11:
That matches yesterday's tune and log

rusefillc, 2022-08-11:
@ElDominio please call when you have a chance

ElDominio, 2022-08-11:
https://user-images.githubusercontent.com/5685108/184249421-e29716b9-266d-4f83-85cc-ffb307757b98.mp4

Video log log5camfirst and log2camfirst tune

ElDominio, 2022-08-11:
https://user-images.githubusercontent.com/5685108/184249510-06843676-243f-41be-8e1a-eb4b94f1de49.mp4

Video of log4cam.log and log1cam.msq

ElDominio, 2022-08-11:
Video of log3cam.log and log1cam.msqhttps://user-images.githubusercontent.com/5685108/184249645-8b0a881c-446f-4500-978d-6474520addcc.mp4

ElDominio, 2022-08-11:
Tune and logs at https://rusefi.com/online/index.php?vehicleName=raymond&user_id=675&sorts[uploadedMsq]=-1&sorts[uploadedLog]=-1

ElDominio, 2022-08-11:
Log1cam.msq is set to second half, log2camfirst is set to first half

rusefillc, 2022-08-11:
this one looks like VVT is happy, decoded intake position is close to zero. around 750 RPM at low RPM car starts to die and maybe as a result of car shutting down intake position goes a bit crazy. 

I suspect that intake VVT here is consequiece not cause of car dieing

https://rusefi.com/online/view.php?log=958

[image]

rusefillc, 2022-08-11:
Separate story https://github.com/rusefi/rusefi/issues/4433

ElDominio, 2022-08-20:
Logs at:

https://rusefi.com/online/view.php?log=985
Can be seen at 2s

https://rusefi.com/online/view.php?log=986
Can also be seen at 2s

mck1117, 2022-08-21:
this is probably the same-ish thing as https://github.com/rusefi/rusefi/issues/4463 and/or https://github.com/rusefi/rusefi/issues/4459

ElDominio, 2022-12-03:
Close

closed 2022-12-03 by rusefillc