rusefi/hellen154hyundai #60: Odd readout in Console, hellen154hyundai set as 60/2

ElDominio, 2022-08-25:
[image]

The car pretty much runs like the picture shows (one cylinder?) 

Log: https://rusefi.com/online/view.php?log=1000

rusefillc, 2022-08-25:
I do not see one cylinder on the picture :( Do you have "show only trigger" or whatever is that checkbox enabled?

ElDominio, 2022-08-25:
"focus on inputs" was selected, my bad

[image]

ElDominio, 2022-08-25:
In individually wired wasted, it runs _better_ but the car still misfires

[image]

mck1117, 2022-08-25:
can you try some old firmware and see if it's different/better/worse?

ElDominio, 2022-08-25:
On july 3

[image]

Same, car is misfiring quite a bit at idle

mck1117, 2022-08-25:
did this car not do this before, but does now? seems like car hardware problem

rusefillc, 2022-08-25:
Sounds like time to https://github.com/rusefi/rusefi/issues/4485

rusefillc, 2022-08-25:
@ElDominio did we ever hard-code a somewhat good tune to at least star a car? or is this one very not OEM?

mck1117, 2022-08-25:
> Sounds like time to [rusefi/rusefi#4485](https://github.com/rusefi/rusefi/issues/4485)

the trigger looks pretty okay to me, but it certainly can't hurt

ElDominio, 2022-08-25:
This car runs perfectly fine on a Haltech PNP though, I just swapped it in and it runs perfectly.

I recorded videos of both but you can only hear crickets lol

mck1117, 2022-08-25:
timing similar? fuel pulse width similar? throttle position similar?

ElDominio, 2022-08-25:
[image]
 for reference
 
 I'll see if i can make it run similarly (this runs seq ignition i know it barely makes a difference)

ElDominio, 2022-08-25:
[image]

WIth these values it's idling extremely lean, even though pulsewidth it .3ms more

mck1117, 2022-08-25:
are we sure the other ECU is running sequential fuel? is it making 2 pulses per cycle or 1?

ElDominio, 2022-08-25:
1 pulse per cylce, can scope it if you need (pnp Haltech comes locked to Sequential anyways)

ElDominio, 2022-08-26:
Idk, after running for a while seems fine, still running individually wired wasted btw

rusefillc, 2022-08-26:
Just pushed global spark and injection counters into logs https://github.com/rusefi/rusefi/commit/cecc5a9d6e7614f94aee6edaaab32fc9a97d1df1 just to have more, more, MORE data in the logs

ElDominio, 2022-12-03:
Close

closed 2022-12-03 by rusefillc