rusefi/hellen154hyundai #62: J2 BOM

rusefillc, 2022-09-01:
row spacing and pitch, somtimes it's square 4.2 sometimes it's not

[image]

rusefillc, 2022-09-01:
C239476 is square would NOT work for us? @Qwerty-OFF 

https://jlcpcb.com/partdetail/238580-C4201WVA2x6P/C239476

rusefillc, 2022-09-01:
@Qwerty-OFF same story C388775 

https://jlcpcb.com/partdetail/XkbConnectivity-X9357WV_2x06HNN2SN/C388775

I am confused. Current PCB wants 5.5 row spacing right?

Qwerty-OFF, 2022-09-01:
C239476:
[image]
C388775:
[image]

> Current PCB wants 5.5 row spacing right?

Yes

closed 2022-11-03 by rusefillc