rusefi/hellen154hyundai #63: Rev C test log

rusefillc, 2022-09-20:
ETB works

rusefillc, 2022-09-20:
@ElDominio have you received any rev C? Do they crank/start real vehicle? Looking to confirm if adjusted crank/cam conditioner is happy.

rusefillc, 2022-09-24:
@ElDominio says "i put on forte, forte turned on without problem"

ElDominio, 2022-09-24:
[DanielRiveraGenesisrevC.zip](https://github.com/rusefi/hellen154hyundai/files/9637739/DanielRiveraGenesisrevC.zip)

No RPM :(
Will upload to rusefi online just in case

https://rusefi.com/online/view.php?log=1052

rusefillc, 2022-09-24:
[image]

Diode goes off first 

https://rusefi.com/docs/ibom/hellen154hyundai-c-ibom.html

ElDominio, 2022-10-05:
[image]

ElDominio, 2022-10-05:
[image]

Qwerty-OFF, 2022-10-05:
Try this
[image]

ElDominio, 2022-10-13:
Did the mod, still no RPM anywhere, TS, console or LED.

ElDominio, 2022-10-13:
[image]

ElDominio, 2022-10-13:
the two on cam were disconnected for the test

i only had 2k resistors to install no 4.7; i normally use 1k pullup so i fugred 2k was closer to 4.7 lol

leds stay off all the time

rusefillc, 2022-10-16:
[image]

rusefillc, 2022-10-16:
@ElDominio looks like part of the issue was software not adjusted to rev C changes

With original "remove six parts" https://github.com/rusefi/hellen154hyundai/issues/63#issuecomment-1256968831

but without "add resistors" I have Hall sensor working on the bench, worth trying on real vehicle with latest firmware.

ElDominio, 2022-10-18:
Updated to latest, still not reading
[image]

rusefillc, 2022-10-19:
All I need is a photo of the board in current state and a log of cranking attempt to confirm if we got any counter activity or not.

ElDominio, 2022-10-19:
Will do tomorrow

ElDominio, 2022-10-20:
[crankingrevc.zip](https://github.com/rusefi/hellen154hyundai/files/9834361/crankingrevc.zip)

rusefillc, 2022-10-20:
@ElDominio can you hack a pull-up to 5v externally or internally?

to begin with I suggest we just focus on Crank

@Qwerty-OFF was a bit wrong at
[image]

in reality 5v is on top side of the other resistor array, the one closer to LEDs. left-bottom corner

[image]

another option is waiting for me to land in PR

rusefillc, 2022-10-20:
actually please remove the wrong resistor lead from

[image]

note how it's connecting things which were not supposed to be connected

ElDominio, 2022-10-20:
STill no RPM

[image]