rusefi/hellen154hyundai #72: Boards with single smiley face

rusefillc, 2022-11-03:
Both boards with single smiley face need capacitor mod

step 1: most of the fix is to remove two capacitors as following

[image]

Later point we will replace 1nF capacitor but there is good hope of progress just that

rusefillc, 2022-11-17:
@ElDominio this would need to be done on all boards

ElDominio, 2022-12-19:
[image]

Single smiley with caps removed, still no CKP :(

rusefillc, 2022-12-19:
I see that https://github.com/rusefi/hellen154hyundai/issues/73 was not applied

[image]

ElDominio, 2023-01-05:
I see that #73 is called "no smiley face"

ElDominio, 2023-01-31:
RevC reading incorrectly incorrectly with fix applied.
[DataLogs.zip](https://github.com/rusefi/hellen154hyundai/files/10551828/DataLogs.zip)

[image]

Kind of reads, but fails to sync and throws trigger error

closed 2023-04-17 by rusefillc