rusefi/hellen154hyundai #80: Please provide happy rev B tune & log

rusefillc, 2023-02-01:
ideally on "20221209_release" but any tune & log of happy vehicle with functional rev B board would work

Ankyth, 2023-02-01:
[Log 2-1-23 RevC fix.zip](https://github.com/rusefi/hellen154hyundai/files/10562867/Log.2-1-23.RevC.fix.zip)

mck1117, 2023-02-01:
capture an SD card tooth log too if you can - that is the ground source of truth that the ECU sees.

rusefillc, 2023-02-13:
Talking to Ray on the phone
Ray says he is looking at 6 boards all with while molex (meaning rev C) in front on him

rusefillc, 2023-02-13:
Ray says four boards have two small capacitors removed, two boards still have small capacitors.

rusefillc, 2023-02-22:
@Timbelom three boards should be in NJ soon

* please write unique IDs on these boards Feb1, Feb2, Feb3

* current assumption is that one board has 4.7K array not soldered well, please add solder
* two boards are in different states of not being modified
* remove caps
* remove diode+transistor, install 0R
* replace resistor array

rusefillc, 2023-02-26:
@Timbelom says five boards have arrived
@Timbelom says two boards do not have #72 fix labeled "f2" and "f3"
@Timbelom says board with most modifications is now labeled "f1"

rusefillc, 2023-02-26:
[image]

Timbelom, 2023-02-26:
F4 solder added

Timbelom, 2023-02-26:
F3 & F2 capacitors removed

Timbelom, 2023-02-26:
F1, F2 & F3 replaced resistor arrays to 4.7K

Timbelom, 2023-02-26:
F2 & F3 removed diode+transistor, installed 0R

rusefillc, 2023-02-27:
F3 & F5 both cam channels  removed diode+transistor, installed 0R

[image]

Ankyth, 2023-03-27:
F3 and F5 work as intended, both read cam and crank.
F2 reads crank but no CAM

rusefillc, 2023-03-27:
Hallelujah! Let me modify two more boards to best spec and ship to PR.

rusefillc, 2023-03-27:
> Hallelujah! Let me modify two more boards to best spec and ship to PR.

https://tools.usps.com/go/TrackConfirmAction?tRef=fullpage&tLc=2&text28777=&tLabels=9405511206214367533690

closed 2023-04-17 by rusefillc