rusefi/hellen154hyundai #88: Migrate schematics to megamodule

rusefillc, 2023-03-28:
we shall migrate schematics from mcu to mega-mcu

Nadezhda-n, 2023-03-29:
Вот эта трансформация меня немного настораживает ( в модулях также приходят на разные ножки mcu, мб это изменения именно alphax-2chan?)
[image]
[image]
@rusefillc

Nadezhda-n, 2023-03-29:
alphax-4 chan у меня не открывается
[image]

rusefillc, 2023-03-29:
> Вот эта трансформация меня немного настораживает ( в модулях также приходят на разные ножки mcu, мб это изменения именно alphax-2chan?)

пока не понимаю о чём ты. там добавляется ВТОРОЙ CAN, там не двигается первый. Уточни пожалуйста вопрос.

сделал тикет https://github.com/andreika-git/hellen-one/issues/277

Nadezhda-n, 2023-03-30:
Полагаю, мне также нужно поменять Module-input-0.1 на Module-vr-discrete-0.4 (судя по внутреннему содержимому модулей это так)
[image]
[image]
@rusefillc прошу подтверждения, чтобы не было самоуправства и путаницы, мало ли:)

rusefillc, 2023-03-30:
LOW_EXT1 & LOW_EXT3 нужно перенести на любые свободные пины

Ага, у нас тут видимо для экспериментов две каншины M6 & M13

Висящие P86 P87 намекают, что M6 нужно просто стереть и P86 P87 стереть пока

[image]

2chan говорит нам, что вторая CAN шина у нас теперь на SPI2
[image]

Nadezhda-n, 2023-03-30:
> LOW_EXT1 & LOW_EXT3 нужно перенести на любые свободные пины
> 
> Ага, у нас тут видимо для экспериментов две каншины M6 & M13
> 
> 2chan говорит нам, что вторая CAN шина у нас теперь на SPI2. значит эту шину просто переносим аналогично.
> 
> [image]
> 
> [image]
по аналогии на SPI2 занято
[image]
Остались только просто IO

Nadezhda-n, 2023-03-31:
[image]
Вот этого разъема только нет в мегамодуле, он не нужен?

Nadezhda-n, 2023-03-31:
Что делать, если не совпадают номиналы подтяжек на плате и внутри мегамодуля?
[image]

rusefillc, 2023-03-31:
> этого разъема только нет в мегамодуле, он не нужен?

J3 убираем, это рудимент какой-то

Nadezhda-n, 2023-03-31:
> 
Отлично! Где/как мне это отобразить?
Вот все подтяжки:
[image]

Nadezhda-n, 2023-03-31:
> > @Nadezhda-n создай плиз отдельный тикет, чтоб вопрос точно не забылся
> 
> @Nadezhda-n создай плиз отдельный тикет, чтоб вопрос точно не забылся

#93 создала, на плате большой тогда это убираю и пока со схематиком тут я все

Nadezhda-n, 2023-03-31:
А вот еще!
Оставить как было или перенести на соответствующие места?
[image]

rusefillc, 2023-04-02:
@Nadezhda-n вопрос очень хорошие, но лучше менять по минимуму - там прошивке нужно быть совместимой, чем меньше различий в железе - тем проще жить

Qwerty-OFF, 2023-04-19:
Будем считать что миграция успешна

closed 2023-04-19 by Qwerty-OFF