rusefi/hellen154hyundai #98: Three hellen154hyundai revC have crank sensor issues

ElDominio, 2023-04-21:
https://rusefi.com/online/view.php?msq=1451

mck1117, 2023-04-21:
qualitative notes about how it is?

rusefillc, 2023-04-21:
@ElDominio @Ankyth in previous episode we had five boards https://github.com/rusefi/hellen154hyundai/issues/80 marked F1, F2, F3, F4 and F5

Are we looking at any of those F boards or else?

https://user-images.githubusercontent.com/48498823/221632325-791a9a6e-fc44-4a88-a83a-d373852ef617.png

Ankyth, 2023-04-21:
Yes, we have board F7 with this issue. the other 2 boards have all the fixes but werent sent to you

ElDominio, 2023-04-21:
Edge and Explorer are screwed up on my PC

We have three revC boards with no discernible difference in hardware with CRANK sensor reading issues. Use the same 4 way compound 4.7k resistor, but the crank signal is incredibly screwed up. Any ideas on what to check  (another revC with same firmware and same tune turns on the car just fine)

ElDominio, 2024-02-23:
what was this? they all seem fine?

rusefillc, 2024-02-23:
I see something was poorly documented a few months ago :(

closed 2024-02-23 by rusefillc