rusefi/hellen154hyundai #102: Error PA8 required by ETB enable but used by ETB disable?

ElDominio, 2023-05-22:
[image]

rusefillc, 2023-05-22:
@ElDominio what board rev? Please link tune

Ankyth, 2023-05-22:
[CurrentTune - Copy.zip](https://github.com/rusefi/hellen154hyundai/files/11537021/CurrentTune.-.Copy.zip)

REV. B

rusefillc, 2023-05-22:
@Ankyth please try again in 29 minutes from now or later

looks like rev C tune loaded into rev B? anyway in 28 minutes this should work better

Ankyth, 2023-05-24:
Updated firmware and its fixed

closed 2023-05-24 by rusefillc

Ankyth, 2023-05-24:
Started the car again, had to update again and with the key on, the throttle dances from 0 to 100 over and over, once i turn the car it goes 100% open, even after calibrating

rusefillc, 2023-05-24:
@Ankyth looks like a separate issue need separate ticket with tune and log

ElDominio, 2023-06-07:
Problem fixed by activating stepper mode and switching h bridge direction