rusefi/hellen154hyundai #103: Throttle goes to 100% when starting car

Ankyth, 2023-05-25:
[throttle not calibrating revC.zip](https://github.com/rusefi/hellen154hyundai/files/11560261/throttle.not.calibrating.revC.zip)

[CurrentTune.zip](https://github.com/rusefi/hellen154hyundai/files/11560267/CurrentTune.zip)

Ankyth, 2023-05-25:
[image]

Throttle wide open while ecu reads 5.7%

Ankyth, 2023-05-25:
@rusefillc

rusefillc, 2023-05-25:
@Ankyth I have a feeling that we are dealing with something weird from C tune loaded into B

like for instance here is the potential issue

[image]

rusefillc, 2023-05-25:
@Ankyth please try with latest firmware in 24 minutes from now - firmware would be auto-adjusting "inverted driver pins" flag based on board revision

Ankyth, 2023-05-27:
Tested and it did not work properly, after calibrating, gauges show like this
[image]

Throttle IS closed fully but it works backwards, i can turn the car on, you press the pedal and does nothing, you release the pedal and it accelerates the amount you pressed the pedal

[CurrentTune (5-27).zip](https://github.com/rusefi/hellen154hyundai/files/11582515/CurrentTune.5-27.zip)

rusefillc, 2023-05-27:
@Ankyth please reset tune and test default

Ankyth, 2023-05-27:
> @Ankyth I have a feeling that we are dealing with something weird from C tune loaded into B
> 
> like for instance here is the potential issue
> 
> [image]

Where is this option?

rusefillc, 2023-05-27:
@Ankyth reminder about top-right corner "Search" button in TS

i really suggest "reset firmware"

Ankyth, 2023-05-27:
Issue fixed, car working as normal

closed 2023-05-27 by Ankyth