rusefi/hellen154hyundai #105: rotating components in CPL revC

Nadezhda-n, 2023-06-14:
name 	case 			rot_cpl	rot_new
TLE9201SG	PG-DSO-12-11_VIA	180	90
ISL9V3040D3S	TO-252-2		180	0

knock:
LMV321IDBVR	SOT-23-5		270	0

power12
MIC5205-3.3YM5	SOT-23-5		270	0

hellen1-input
SRV05-4-P-T7	SOT-26			0	270

MCU
SRV05-4-P-T7	SOT-26			90	0

Nadezhda-n, 2023-06-15:
[1111.txt](https://github.com/rusefi/hellen154hyundai/files/11756141/1111.txt)

rusefillc, 2023-06-15:
https://github.com/andreika-git/hellen-one/issues/302

Nadezhda-n, 2023-06-15:
1 есть
[image]

[image]
Было/стало