rusefi/hellen154hyundai #106: Why is there a 12pin connector with only 5 pins in use?

ElDominio, 2023-06-27:
Is there no i/o left for more?

Haltech has
2x analog
2x digital
4x output

on their connector, do we only have enough for the current setup? or are there undocumented pins?

ElDominio, 2023-06-27:
[image]

I see it was changed in firmware, but there is no user facing mention of this; where are they on the connector?

rusefillc, 2023-06-27:
@ElDominio remember how snapshot takes some time to build? Same thing with pinout - it takes some time to build once file is updated.

https://rusefi.com/docs/pinouts/hellen/hellen154hyundai/?pin=X11

one pin is a separate issue https://github.com/rusefi/hellen154hyundai/issues/107

closed 2023-06-27 by rusefillc