rusefi/hellen154hyundai #111: rev c: uneven R from CAM to +5v

rusefillc, 2023-06-30:
latest batch of rev C boards

two boards, both completely offline
from J20 (+5v test pad) to pad 1 of D40 (CAM1) 2.1K
from J20 (+5v test pad) to pad 1 of D41 (CAM2) 3.8K
from J20 (+5v test pad) to pad 1 of D44 (CRANK) 3.8K

Question: feature or bug? I would assume same resistance at lest between CAM1 and CAM2?

rusefillc, 2023-06-30:
[image]

Qwerty-OFF, 2023-06-30:
Удали R72, везде должно стать одинаково 3.8
Удали три нулевика, везде должно стать 4.7
Зачем там стоят нулевики я не знаю, их нет в BOM. Возможно это какая-то странная попытка починить светодиоды, которые в этой ревизии ломали датчики холла

rusefillc, 2023-06-30:
R72 removed - all three measure 3.8K now

rusefillc, 2023-06-30:
в ревизии D нужно будет проверить все подтяжки у холлов

closed 2024-03-21 by Qwerty-OFF