rusefi/hellen154hyundai #113: Base tune for 154hyundai

ElDominio, 2023-07-02:
Should get most key start Hyundai Genesis Coupe started

If push button start or immobiilized, need to add +360 to trigger angle

https://rusefi.com/online/view.php?msq=1498

ElDominio, 2023-07-18:
Better BK2 base:
https://rusefi.com/online/view.php?msq=1507

Better BK1 will be uploaded shortly, almost ready

ElDominio, 2023-07-18:
https://rusefi.com/online/view.php?msq=1508

ElDominio, 2023-11-25:
How much effort to make the last linked tune the base tune for 154?

ElDominio, 2023-11-25:
> Better BK2 base:
> https://rusefi.com/online/view.php?msq=1507
> 
> Better BK1 will be uploaded shortly, almost ready

How much effort to make the this tune the base tune for 154?

rusefillc, 2023-11-27:
today BK2 https://rusefi.com/online/view.php?msq=1578

comparison https://rusefi.com/online/diff.php?msq1=1578&msq2=1507

@ElDominio need a new BK2 default based on more recent firmware

ElDominio, 2024-02-23:
Do we still need this?