rusefi/hellen154hyundai #115: Possible user error led to smoke

ElDominio, 2023-07-20:
User had a car with hellen154
Fans stopped working
Unclear things happened
Mysteriously ECU burned a coil driver, all injector/ignition/relays latched on, flooding cylinders and burning out coil driver, probably blew a fuse

My theory is user tried powering the CAR's 12v rail with ECU off (no 12v key), resulting in powering of everything on the rail since natural state of unpowered ECU is grounding all outputs.

Is my theory reasonable?
[image]

ElDominio, 2023-07-20:
Best way to reproduce in my opinion: 

Apply ground to proper ground pins
Apply 12v to outputs which drives 12v solenoids like fan output pin, vvt output pin, ignition output pin (instead of correct 12v input)

ElDominio, 2023-07-30:
ECU was tested and even though the igniter looks burnt, coils fire. Wtf? ECU seems fine

rusefillc, 2023-07-30:
@Qwerty-OFF ^^^ 

See also https://github.com/rusefi/stim/issues/20

ElDominio, 2023-07-30:
I don't think so, since it's a plug and play unit and the car was running already.

If the ECU is not powered (but connected), and say the user puts 12v to the Fan Relay output, what would happen? In my previous experiences this usually allows the ECU to power up without being fed in the correct way (won't see 12v in the key in pin) and most outputs provide ground.

Sadly, I am also terrible at electronics lol

ElDominio, 2023-08-11:
The ECU definitely works fine. It has a burn mark on IGBT but it fires the coil just fine and is not burst or inflated or anything. Closing?

closed 2023-08-11 by ElDominio

rusefillc, 2023-08-11:
@Qwerty-OFF @dron0gus how could we have signs of heat next to IGBT yet functioning IGBT? is it that PCB is weaker than IGBT itself?

mck1117, 2023-08-11:
is there no heatsinking planes/vias for those igbts?