rusefi/hellen154hyundai #119: rev C PPS2 voltage

rusefillc, 2023-09-30:
rev C on bench
using stim reference pin 11B which documentation says should be 1.1v

 ***********************************************
 * BAD channel 3 (PPS2): voltage 0.980392192 (raw 50) not in range (1.017500042..1.182500128)
 ***********************************************

rusefillc, 2023-09-30:
Second board same rev C

 ***********************************************
 * BAD channel 3 (PPS2): voltage 0.960784384 (raw 49) not in range (1.017500042..1.182500128)
 ***********************************************

ElDominio, 2024-06-02:
I have two more with this same issue, i am assuming this was a batch batch and some got sent out, is there a know fix?

ElDominio, 2024-06-02:
[image]
 
this hole is next to R126 but iBom does not show that even exists? there are traces that run to those pads?

rusefillc, 2024-06-02:
closing issue because we have already fucked things up while unrelated issues.

closed 2024-06-02 by rusefillc