rusefi/hellen154hyundai #126: Aux LS1, X8 not working

ElDominio, 2023-12-05:
[image]

I have a boost controller solenoid wired to Aux LS1 and it is not responding.  Seems to me PB3 is used to trigger it?

rusefillc, 2023-12-07:
[image]
rev C
[image]

rusefillc, 2023-12-07:
``#define H_SPI1_SCK Gpio::B3``?

rusefillc, 2023-12-07:
fixed

```
16/29 MIL GOOD channel 17
               Setting ADDR 1
               sending line=18 value=1
                      ADC says HIGH 18@0 3.065v
               sending line=18 value=0
                      ADC says LOW 18@1 0.169v
               sending line=18 value=1
               sending line=18 value=0
17/29 Tach GOOD channel 18
               Setting ADDR 2
               sending line=19 value=1
                      ADC says HIGH 19@0 3.066v
               sending line=19 value=0
                      ADC says LOW 19@1 0.003v
               sending line=19 value=1
               sending line=19 value=0
18/29 X8 AuxLS1 GOOD channel 19
```