rusefi/hellen154hyundai #131: X9 Digital 1 as Flex Fuel input shows 0% with 60hz

ElDominio, 2023-12-27:
I want to read flex fuel %, and I have connected a GM flex fuel sensor to pin X9. I removed the 680k pulldown, added a 1k pullup at R14, and measured the flex sensor output, which reports 60hz on an frequency multimeter. For some reason TS shows 0% flex fuel. Log and tune attached here: 
[flexlog.zip](https://github.com/rusefi/hellen154hyundai/files/13781625/flexlog.zip)

rusefillc, 2023-12-27:
@ElDominio please use ``flexinfo`` command in console

ElDominio, 2023-12-28:
2023-12-28_15_21_43_851: EngineState: flex counter 0
2023-12-28_15_21_43_852: EngineState: lowFlexCallbackCounter counter 0
2023-12-28_15_21_43_852: EngineState: flex freq 0.000000000
2023-12-28_15_21_43_852: EngineState: pulseWidthUs 0.000000000
2023-12-28_15_21_43_852: EngineState: latestCallbackTime 536893112
2023-12-28_15_21_43_852: EngineState: confirmation_flexinfo:8

ElDominio, 2023-12-28:
[image]

rusefillc, 2023-12-31:
@ElDominio wrong resistors were touched?

we have never populated modernized rev D which you might be looking at https://github.com/rusefi/hellen154hyundai/tree/main/boards/hellen154hyundai-d

you have rev C board which visually clearly does not match the screenshot you've provided. Please share photo of REAL modified board.

ElDominio, 2024-01-02:
[image]

rusefillc, 2024-01-04:
[image]

[image]

rusefillc, 2024-01-04:
@ElDominio problem 
#74HC2G17GW,125
U490,
U491,

are not populated on all boards :(

ElDominio, 2024-01-04:
They are buffers? Can i just jump pads?

rusefillc, 2024-01-04:
```
2024-01-04_02_01_31_580: EngineState: flex counter 3708
2024-01-04_02_01_31_581: EngineState: lowFlexCallbackCounter counter 1854
2024-01-04_02_01_31_581: EngineState: flex freq 50.005001068
2024-01-04_02_01_31_581: EngineState: pulseWidthUs 17445.000000000
2024-01-04_02_01_31_586: EngineState: latestCallbackTime 536892784
```

rusefillc, 2024-01-04:
yes buffers
yes approved to jump

X9 digital 1 are the two top pads

[image]

ElDominio, 2024-01-05:
fix works and we have flex

rusefillc, 2024-01-05:
@Qwerty-OFF please manually fix rev C BOM since there is a chance of rec C re-order before we go rev D

Qwerty-OFF, 2024-02-21:
Now, the function of the dinput module is performed by the mega-module, buffers will be installed

closed 2024-02-21 by Qwerty-OFF

ElDominio, 2024-12-13:
> @ElDominio problem #74HC2G17GW,125 U490, U491,
> 
> are not populated on all boards :(

i hate 74HC2G17GW fun fact

ElDominio, 2025-05-05:
> > [@ElDominio](https://github.com/ElDominio) problem #74HC2G17GW,125 U490, U491,
> > are not populated on all boards :(
> 
> i hate 74HC2G17GW fun fact

i now love 74HC2G17GW  and i was very dumb for saying this