rusefi/hellen154hyundai #137: WBO module: 2x3 moles and maybe OEM header as well

rusefillc, 2024-02-23:
BK2 2.0T has WBO stock

rusefillc, 2024-02-23:
@dron0gus which WBO pin is which? is that enough info to decode?

[image]
[image]

[image]

dron0gus, 2024-02-24:
Does not look like default LSU4.9 pinout.
[image]
Do we know for sure that this is Bosch LSU?

Qwerty-OFF, 2024-03-21:
[image]