rusefi/hellen154hyundai #141: C67 CPSV needs to be controlled to not have issues with fuel tank depressurization

ElDominio, 2024-04-23:
On most cars, this is still present. Not having it controlled casuses vacuum to build up in the fuel tank, and once the engine is off, the tank has a pressure release that sounds very sharp and frightens most owners. No bad things happen except for the noise but people get scared so this should be controlled (just a normal LS)

Are there pins left for this?

rusefillc, 2024-04-25:
@ElDominio I am not familiar with CPSV acronym can you please spell out what valve is that?

ElDominio, 2024-04-25:
EVAP Purge

rusefillc, 2024-04-25:
https://github.com/rusefi/hellen154hyundai/issues/142 at least to name on the schematics

rusefillc, 2024-04-25:
@ElDominio I see we have P12 on rev C and rec D

What is the ask here? For rev C and rev D you would have to jump wire from P12 to one of aux LS

in later revision we can make an official separate pin

ElDominio, 2024-04-26:
Then do this please

rusefillc, 2024-05-13:
Will be handled by https://github.com/rusefi/hellen154hyundai/issues/142

closed 2024-05-13 by rusefillc