rusefi/hellen154hyundai #143: Connect onboard wideband to stock wideband pinout present ONLY ON BK2

ElDominio, 2024-05-07:
BK2 (2013+ Genesis Coupes) very probably have onboard wideband. Can we connect to this _in addition to_ the connector on board 
[image]

Stock sensor seems to be 6 wire wideband

[image]

4 of the pins

[image]

htr pin

[image]

gets power from a junction box

Now, BK1 uses a narrowband, is it conflicting?
[image]

rusefillc, 2024-05-07:
@ElDominio it's a bit premature to wire things until we know the kind of sensor

@ElDominio please start with OEM sensor part number and good photo

@ElDominio note that we already have doubts at https://github.com/rusefi/hellen154hyundai/issues/137#issuecomment-1962405182

ElDominio, 2024-05-07:
[image]

OEM Repair manual states this information about O2 sensor

rusefillc, 2024-05-07:
* red flag one: sensor pinout does not match https://github.com/rusefi/rusefi/wiki/WBO#naming-convention

* red flag two: does not look like 4.9 to me
 https://www.rockauto.com/en/catalog/hyundai,2014,genesis+coupe,2.0l+l4+turbocharged,3307646,exhaust+&+emission,oxygen+(o2)+sensor,5132 see "upstream"

[image]

[image]

ElDominio, 2024-05-07:
[image]

[image]

[image]

rusefillc, 2024-05-07:
@ElDominio I assume not 4.9 until we have evidence of otherwise

looks like we cannot repurpose harness since calibration resistor is in a different position?

rusefillc, 2024-05-07:
> does having the sensor on hand help? how can this be moved forward; is there a wideband expert we can ping

I believe that we are not in the position to move forward here. We do not have the brains to figure out what sensor this is, and we do not have motivated engineers even if we find out that this is NTK or something.

ElDominio, 2024-05-07:
I will wire one to a wbo2 module and see if it works or it catches on fire and will report back

rusefillc, 2024-05-08:
@ElDominio https://www.haltech.com/news-events/new-release-ntk-o2-wideband-kits/ hints at no chance of success

rusefillc, 2024-07-08:
looks like those are called ZFAS-D, ZFAS-U2 and ZFASU3?

rusefillc, 2024-12-31:
https://www.etas.com/download-center-files/products_ES600/etas-es63x.1-user-guide-en-202110.pdf

[image]

rusefillc, 2024-12-31:
The Bosch lambda sensor LSU 5.1 is a planar single-cell limiting current sensor,
the Bosch lambda sensors LSU 4.2, LSU 4.9, LSU 5.2, LSU ADV, and the NTK
lambda sensor ZFAS-U2 and ZFAS-U3 are planar two-cell, limiting current sensors.
The LSU 4.2 compares the oxygen content of the exhaust gas with the ambient
air. All other broadband lambda probes compare the oxygen content of the
exhaust gas with their integrated oxygen reservoir.
The single-cell probe LSU 5.1 is particularly suitable for use in the lean environment.
As the above-mentioned two-cell sensors consist of a combination of a Nernst
concentration cell (sensor cell) and a pump cell transporting oxygen ions, they
can carry out precise measurements not only in the stoichiometric point at  =
1, but also in the lean and rich ranges.
With the exception of the LSU ADV-D, all sensors are calibrated individually at
the factory.
The operational state of the lambda sensor is characterized by the following
parameters:
• Ri (internal resistance of the lambda sensor)
• Ip (pump current of the lambda sensor)

rusefillc, 2024-12-31:
[image]

rusefillc, 2025-01-04:
[image]