rusefi/hellen154hyundai #152: RevC not reading PPS2

ElDominio, 2024-06-02:
I have two revC h154 that pps2 stays at 0.5v regardless of pedal position. 

PPS1 reads fine, but pps2 no matter the pedal position ALWAYS reads 0.5xV. This is both on tunerstudio and in reality with a multimeter.

rusefillc, 2024-06-02:
@ElDominio with vehicle? with ignition? without vehicle? without ignition? Do you have a known good rev C to compare in same scenario?

ElDominio, 2024-06-02:
With vehicle with ignition; i know many revC out there but these two have this issue

rusefillc, 2024-06-02:
@ElDominio please confirm reading without vehicle, with ECU header completely disconnected

ElDominio, 2024-06-02:
Reading on pin K35?

rusefillc, 2024-06-02:
looking at TS "raw pps2" gauge

how do you read 0.5v? i assume you confirm with TS raw pps2 gauge?

ElDominio, 2024-06-02:
> looking at TS "raw pps2" gauge
> 
> how do you read 0.5v? i assume you confirm with TS raw pps2 gauge?

Yes, both TS "rawPPS2" and multimeter show same 0.5v voltage

rusefillc, 2024-06-02:
@ElDominio please confirm TS raw pps2 reading without vehicle, with ECU header completely disconnected

ElDominio, 2024-06-02:
"pin 95 is giving me zero on my multimeter and .002 on tunerstudio" with ECU header completley disconnected

ElDominio, 2024-06-02:
[image]

rusefillc, 2024-06-02:
in NJ those pads are empty and shiny, why not shiny on the photo above?

[image]

ElDominio, 2024-06-02:
[image]

pads are present, potato cam issue

ElDominio, 2024-06-14:
ECU #1 had bad soldering on pin 73, PS2 gnd. Costa Rica ECU will check for similar issue

rusefillc, 2024-06-14:
@ElDominio next batch would be China soldered instead of NJ weak human soldered

closed 2025-01-04 by ElDominio