rusefi/hellen154hyundai #154: rev E fab

rusefillc, 2024-06-06:
@Qwerty-OFF please confirm orientation

Qwerty-OFF, 2024-06-07:
Выглядит хорошо

rusefillc, 2024-06-29:
хочется опять заказать

* предлагаю C507072 конденсатор https://github.com/rusefi/hellen154hyundai/commit/744f4868d3816b333b33b7bf371355e2c4c5e0d5
* что положить на полку D102 и F410?
* U1 заменил на C377942 https://github.com/rusefi/hellen154hyundai/commit/c2754750d9635b054add0c03cb850d033d637d70

[image]

Qwerty-OFF, 2024-07-01:
C603 поменял на C106862 который мы уже когда-то ставили на замену раньше
D102 ты кажется уже выбрал подходящий C6454456
F410 C369160
U1 ок

rusefillc, 2024-07-09:
@Qwerty-OFF нужна замена C293505 им не хватило одной штуки

Qwerty-OFF, 2024-07-09:
C388775 вписал в BOM