rusefi/hellen154hyundai #161: F7 H154 fan broken again

ElDominio, 2024-08-25:
Same symptoms as harely conflict on F4, where coolant temp settings were being ignored

ElDominio, 2024-08-25:
Ref #140

rusefillc, 2024-08-25:
[image]

rusefillc, 2024-08-25:
yaml looks to match kicad

```
  - pin: K65
    meta: H144_OUT_IO7
    class: outputs
    ts_name: Fan Relay Low
    function: Fan relay low
```

```
  - pin: K88
    meta: H144_OUT_IO8
    class: outputs
    ts_name: Fan Relay HI
    function: Fan relay high
```

rusefillc, 2024-08-25:
K65 and K88 are part of HW QC

```
static Gpio OUTPUTS[] = {
	Gpio::H144_OUT_IO7, // Fan Relay Low
Gpio::H144_OUT_IO8, // Fan Relay HI
```

rusefillc, 2024-08-25:
@ElDominio radiator fan control is a composite behavior which consists of CLT decoding, tune and actual fan output. Please see if you can isolate the problem

@ElDominio do you have "bench test" buttons? @ElDominio do you have "bench test" spinning either of the fans?

rusefillc, 2024-08-25:
PS: what the hell is "harely" other that part of the signal that this ticket was not articulated properly?

ElDominio, 2024-08-25:
Your help was quite helpful

Dealer had set disable fans after 80kmh, but car has VSS error reporting 357kmh. Fans never engaged

rusefillc, 2024-08-25:
@ElDominio let's not close ticket yet, let's see how can we improve TS side

ElDominio, 2025-01-04:
yes?