rusefi/hellen154hyundai #163: When brake is pressed, MAP is incorrect. Disconnecting brake switch fixes the issue

ElDominio, 2024-10-31:
As title states, seems brake switch input is mucking MAP.

https://youtu.be/s4iHTNE2S2A

rusefillc, 2024-10-31:
@ElDominio please provide logs with vehicle not running and tapping on the brake pedal, need to see how MAP sensor voltage changes

ElDominio, 2024-11-01:
[brakelog.zip](https://github.com/user-attachments/files/17601698/brakelog.zip)

raw voltage changes from 1.8 to 2.1

rusefillc, 2024-11-01:
@ElDominio weird 

One board or many boards? One vehicle or multiple vehicles? Only hyundai or any other models?

ElDominio, 2024-11-01:
Only H154, 2x of them. Unaware of rev but definitely on-board wideband

Have not tested vehicles with hellen PNP's with working brake switches

rusefillc, 2024-11-05:
@ElDominio please use DMM

* 5v MAP supply
* MAP signal

ElDominio, 2024-12-08:
can we check if anything about this circuit changed from revD to revC? This is starting to become more of a problem as more units go out

rusefillc, 2024-12-08:
@ElDominio revD is a complete re-work of the board, everything has changed between those revisions

@ElDominio did you have a chance to use DMM on revD board?

ElDominio, 2024-12-08:
@rusefillc  do i need to be able to press brake pedal? I have a Genesis with revD, but brake wire has been cut. If we only need 5v and map signal, I can do that, but brake does not affect this car

ElDominio, 2024-12-08:
Reports:
Brake off:
MAP signal - 0.1v
5v MAP supply - 4.9v

Brake on
MAP signal - 5.3v
5v MAP supply - 4.9v

disconnecting brake input from ECU fixes the issue

rusefillc, 2024-12-08:
@Qwerty-OFF ^^^

rusefillc, 2024-12-13:
@Qwerty-OFF https://github.com/rusefi/rusefi_documentation/tree/master/OEM-Docs/Hyundai "2010 2.0 coupe" and "2014 * 2.0 coupe"

Qwerty-OFF, 2024-12-13:
@ElDominio please measure the following values with a multimeter:

|            | No Brake | Brake |
|------------|----------|-------|
| GND - GNDA |          |       |
| GND - 5V   |          |       |
| GND - MAP  |          |       |

[image]

rusefillc, 2024-12-13:
https://rusefi.com/docs/pinouts/hellen/hellen154hyundai_revD/?pin=K14 agnd
https://rusefi.com/docs/pinouts/hellen/hellen154hyundai_revD/?pin=K31 MAP
https://rusefi.com/docs/pinouts/hellen/hellen154hyundai_revD/?pin=K30 5v

rusefillc, 2024-12-13:
@Qwerty-OFF @ElDominio I am able to reproduce this on a bench - note 0.3v diff in voltage

https://rusefi.com/docs/pinouts/hellen/hellen154hyundai_revD/?pin=A29 floating

[image]

https://rusefi.com/docs/pinouts/hellen/hellen154hyundai_revD/?pin=A29 goes to +12v

[image]

rusefillc, 2024-12-14:
@Qwerty-OFF 

|            | No Brake | Brake |
|------------|----------|-------|
| GND - GNDA |     0     |    0   |
| GND - 5V   |  4.9        |    4.9   |
| GND - MAP  |     4.2     |    4.1-4.2   |

Qwerty-OFF, 2024-12-14:
[image]
[image]

rusefillc, 2024-12-14:
no brake: in 4.2, out 4.2

@Qwerty-OFF 

with brake: in 4.6, out 4.6

rusefillc, 2024-12-14:
brake and MAP share srv05

rusefillc, 2024-12-14:
rev E

https://jlcpcb.com/partdetail/Msksemi-SRV05_4_PT7/C6454456
[image]

rusefillc, 2024-12-14:
rev B zero voltage leakage

[image]

[image]

rusefillc, 2024-12-14:
@Qwerty-OFF please provide perfect digikey/mouser PN

rusefillc, 2024-12-14:
problem: there is another variation of S5 which while better than V05 is not perfect

[image]

Qwerty-OFF, 2024-12-14:
https://www.mouser.com/ProductDetail/Semtech/SRV05-4.TCT?qs=rBWM4%252BvDhIe0tfPjqKjR5Q%3D%3D

rusefillc, 2024-12-17:
@Qwerty-OFF а можно картинку, где мне искать 5в на мегамодуле, они же другие?

Qwerty-OFF, 2024-12-17:
[image]

ElDominio, 2025-04-19:
has the BOM been updated for this or anything?

rusefillc, 2025-04-19:
Ticker is open. That's opposite of closed. Closed would be resolved, open means unresolved.

No fabrication was done since we're discovered, I am not aware of plans to fab. Thus no rush to spend resources of proper fix.