rusefi/hellen154hyundai #166: Clutch switch input A43 does not work

ElDominio, 2024-12-09:
There is no change on two RevE units. We swapped in a RevC and it did work.

RevC log:
[revcClutchswitchWorking.zip](https://github.com/user-attachments/files/18064147/revcClutchswitchWorking.zip)

RevE log:
[revEnonworkingClutch.zip](https://github.com/user-attachments/files/18064167/revEnonworkingClutch.zip)

I think this is only a processor pin issue, but I cannot lua test since the schematics I have don't show MCU :(

ElDominio, 2024-12-09:
[image]

Does not produce any results

rusefillc, 2024-12-12:
@Qwerty-OFF looking for picture of PD/PU?

rusefillc, 2024-12-12:
```
#RES2 CLUTCH: DOWN 680k 100n
R240,
R244,"680k","R0402","C163456"
C219,"100n","C0402","C307331"
```

Qwerty-OFF, 2024-12-12:
To configure pullup you need to move the resistor
[image]
If you can't move the small resistor, you can try adding 10k this way
[image]

Qwerty-OFF, 2024-12-13:
Fixed in the replacements of revision E

closed 2024-12-13 by Qwerty-OFF