rusefi/hellen154hyundai #173: Cannot assign wastegate position on revC

ElDominio, 2025-05-09:
[image]

I see schematic says wastegate sensor pin is PA5?

rusefillc, 2025-05-09:
@ElDominio lack of tune makes me assume user error

ElDominio, 2025-05-09:
[cantwastegate.zip](https://github.com/user-attachments/files/20125638/cantwastegate.zip)

ElDominio, 2025-05-09:
[image]

Should main.yaml on non-F7 154 say H144_IN_AUX4_DIGITAL instead of H144_IN_AUX4 ?

[image]

rusefillc, 2025-05-09:
@ElDominio looks like 4chan has similar issue

```
  - pin: E7
    meta: H144_IN_D_AUX4
    class: event_inputs
    ts_name: E7 - Flex
    function: Digital Input / Flex fuel sensor
    color: brown
```

ElDominio, 2025-05-09:
Tested changing H144_IN_D_AUX4 to H144_IN_AUX4_DIGITAL , issue is NOT fixed on my LOCAL BUILD which does not have all the changes on 5ed8585 , will test commit hanges now

ElDominio, 2025-05-09:
[image]

The _D_ was left, but we don't want the _D_

closed 2025-05-09 by ElDominio