rusefi/hellen154hyundai #174: Cannot assign wastegate position sensor on RevC

ElDominio, 2025-05-09:
[image]

I am trying to set Analog wastegate position sensor to use ETB-style wastegate actuator (which requires this analog input)

When this is set, the error shown on the screenshot shows up.

ElDominio, 2025-05-09:
[cantwastegate2.zip](https://github.com/user-attachments/files/20127025/cantwastegate2.zip)

rusefillc, 2025-05-09:
[image]

rusefillc, 2025-05-09:
@Qwerty-OFF 

```
// AIN20 PA7
#define H144_IN_AUX3_ANALOG EFI_ADC_7
#define H144_IN_AUX3_DIGITAL A7
// it's a mess see also H144_IN_D_AUX4
// AIN22 PC5
#define H144_IN_AUX4_ANALOG EFI_ADC_15
#define H144_IN_AUX4_DIGITAL C5
```

vs
```
// A22
#define H144_IN_D_AUX3 C5
#define H144_IN_D_AUX3_ANALOG EFI_ADC_15
// it's a MESS see also H144_IN_AUX4_DIGITAL
// A20 AIN20
#define H144_IN_D_AUX4 A7
```

looks like aux3 and aux4 are in different order? which one is what?

Qwerty-OFF, 2025-05-09:
https://github.com/andreika-git/hellen-one/issues/356
A little confusion in the scheme that has become our new standard in megamodules

rusefillc, 2025-05-10:
@ElDominio I believe that the mess was significantly reduced, and the issue should be fixed now

ElDominio, 2025-05-15:
fixed! yay!

closed 2025-05-15 by ElDominio